markqvist/NomadNet #1: Unable to use UI

whitecd, 2021-05-26:
I have installed and configured NomadNet using "pip3 install nomadnet" on two different PCs.  Installation of RNS, LXMF, and NomadNet appear to be correct.  After execution of 'nomadnet' (Log = 7), I get the following terminal stream and nothing else:
"nomadnet 
[2021-05-26 17:53:23] [Notice] Configuration loaded from /home/curt/.reticulum/config
[2021-05-26 17:53:23] [Debug] Started shared instance interface: Shared Instance [37428]
[2021-05-26 17:53:23] [Notice] Skipping disabled interface "TCP Server Interface"
[2021-05-26 17:53:23] [Notice] Skipping disabled interface "TCP Client Interface"
[2021-05-26 17:53:23] [Notice] Skipping disabled interface "RNode LoRa Interface"
[2021-05-26 17:53:23] [Notice] Skipping disabled interface "Packet Radio KISS Interface"
[2021-05-26 17:53:23] [Notice] Skipping disabled interface "Packet Radio AX.25 KISS Interface"
[2021-05-26 17:53:23] [Verbose] Loaded 4 known destination from storage
[2021-05-26 17:53:23] [Verbose] Loaded Transport Identity from storage"

I was expecting the UI to appear, but it does not.  Although documented in the READ.ME file, is the UI implemented in Nomad Network Client 0.0.5?  Any help would be greatly appreciated.
Thanks for your time and consideration.

markqvist, 2021-05-27:
Thanks for reporting this issue.

I looked into this a bit, and it seems that something funky is going on with the 88-colour mode that nomadnet tries to use in some terminals. Are you using Konsole as your terminal emulator by chance?

Can you try changing to 16-colour mode in the configuration file? Just comment out the 88-colour mode line and enable the 16-colour one. Please let me know if it works. 

Alternatively, I'd also be interested in knowing if the higher colour modes work if you run nomadnet inside of tmux, which interestingly enough seems to alleviate the issue on my end. 

I will be reverting to 16 colours as the default in the next preview release, and let the user set a higher colour mode if wanted. It will probably be more compatible.

whitecd, 2021-05-27:
Thanks for the rapid reply.
Yes, I am using Konsole as my terminal window.

As to changing settings in the configuration file.  I find no default color variables in $Home/reticulum/config.  Also, I can not find any references to a NomadNet configuration file, either on my system, nor in the Git files.   Could you assist me with a path?

I am able to visualize the NomadNet interface using TMUX. I am to see the Announcements on one PC, but not on the other.  I've opened the firewall to pass 37428 and 4242, but not changes.  I've also enabled transport on both PCs, but still only one-way recognition of announcements.

Thank you for your support.

markqvist, 2021-05-27:
Thanks, good to know. I also found the problem to happen in Konsole. To resolve this, for now I will set the default color mode to 16 colors to be more compatible.

The configuration file is in $home/.nomadnet/config. Under the UI section, there is an option for color mode.

Are the PCs in the same subnet/broadcast domain? Or are they on different subnets? As in, are they on the same physical ethernet / wifi network, or on different networks?

whitecd, 2021-05-27:
Found config- Thanks
16 color mode functions with Konsole terminal.

Currently turned off UFW and a TUN-based P2P Virtual Network application.  Each PC is connected to both a wired and wireless network.  Each network is in a different subnet, but each PC are in the same respective subnets. Also, removed all files/directories in ~/.nomadnetwork/storage to remove persistence.
 
NomadNet recognizes announcements from its respective peer and allows me to hand create matching conversations.  I can now pass messages from one PC to the other.

Currently there are no known nodes and no remote nodes, even though they are 'defined' in the Conversation window.  Are 'known nodes'  manually assigned in the  'peersettings' file?

Thanks again for the assistance.

markqvist, 2021-06-29:
The node functionality is not yet implemented, but I'm working on it now. All participants on the network are thought of as peers and can communicate with each other directly or over multiple hops. But you can also elect to host a node, which will allow other users to connect to your node and browse pages, download files and resources or interact with external programs.

markqvist, 2021-07-06:
You can now (in version 0.0.6) create conversations directly from the announce stream, so you don't have to manually type in destination hashes anymore.

More conservative color mode has been set as default, and the in-program guide details that colors should be set higher if possible.

Node hosting and therefore node discovery, connecting to and interacting with nodes is now next on the implementation list :)

closed 2021-07-06 by markqvist