markqvist/NomadNet #2: Selecting what interface to use in config file

SePcGuy, 2021-07-02:
Not sure if to post this to either the Reticulum repository, or this one. Anyways, I want to use UDP broadcast to set up a network of nodes with NomadNet. However, all of my nodes have more than one interface. The first interface being how I ssh into the machines/how they connect to the internet. The machines all then have an extra interface (eth1) that operates as a LAN between them. Is there a way in the .reticulum/config file to select the interface to use the default UDP broadcast on?

I tried looking at the code to see what the deal is, but I'm in over my head.

Thanks.

markqvist, 2021-07-02:
It is currently possible to listen on only one interface, but I'd agree it's not very obvious, and not in the most elegant way possible. The default UDP interface has "listen_ip" set to 0.0.0.0 which will listen on _all_ interfaces. If you know the IP address of your eth1 interface, you can specify that as listen_ip. You can also configure forward_ip to the broadcast address of only the eth1 interface, and Reticulum will only broadcast on that interface.

This can be a problem though, if you don't know the IP addresses beforehand - for example when they are DHCP assigned.

I will implement a more user-friendly way of doing this. It should be possible to just specify an interface name, and have Reticulum listen and broadcast on all IPs associated with that interface. I'll probably have this included in the next beta release of Reticulum.

Hope you can make do with manually specifying the listen_ip until then.

SePcGuy, 2021-07-06:
Thank you for the comment. I tried setting listen_ip to the second interface's IP address, but it didn't work while doing that in any case. However, I was able to get it working by leaving listen_ip as 0.0.0.0, but setting forward_ip to the broadcast address of the second interface's subnet. Thanks for the help!

markqvist, 2021-07-06:
Ah, interesting. Thanks for the feedback on that. It definitely needs some reworking, but since it's really related to Reticulum, I'm going to close this issue and open a corresponding one on the Reticulum repository. You're welcome to subscribe to that issue as well if you want updates. I will probably have this fixed in the next beta release of Reticulum.

Also, you might want to check out Nomad Network 0.0.6 which I just released, it adds quite a few ease-of-use improvements.

markqvist, 2021-07-06:
Link to relevant Reticulum issue: https://github.com/markqvist/Reticulum/issues/5

closed 2021-07-06 by markqvist