markqvist/NomadNet #7: Exec format error while processing page request

HarlekinSimplex, 2022-06-28:
Hi Mark,

actually I am using a custom location for the nomad config and storage files.
While processing a request for the index.mu page I see the following error in the log:

```
[Verbose] Incoming link request <39890900244f18b6d778> accepted
[Verbose] Peer connected to <nomadnetwork.node.436796059edc9fa3684c/eeda32d48d3603a6c16e>
[Debug] Handling request <a124d8f4efadfe22807f> for: /page/index.mu
[Verbose] Page request <a124d8f4efadfe22807f> for: /page/index.mu
[Verbose] Serving page: /home/bsb/.nomadnetwork/paulchen/storage/pages/index.mu
[Error] Error occurred while handling request <a124d8f4efadfe22807f> for: /page/index.mu
[Error] The contained exception was: [Errno 8] Exec format error: '/home/bsb/.nomadnetwork/paulchen/storage/pages/index.mu'
```

The file is there and can be read.
Renaming it gives the info that the node is serving pages but index.mu is not yet available.
It contains nothing else than just 'HelloWorld' and is stored as a UTF-8 file.

Any clue what might be wrong here causing that error.
I did already run out of ideas what to try...

Cheers
Stephan

markqvist, 2022-06-28:
It looks like nomadnet is trying to execute the page to generate a dynamic response. Did you set the executable permission on the file? If the file is marked as an executable, nomadnet will treat it as such, and return whatever *output* the execution generates, instead of it contents.

But if your file is not actually executable in any way, it would generate an error like this.

markqvist, 2022-06-28:
Good to hear it was as simple as that! I will just close this issue then :)

closed 2022-06-28 by markqvist

HarlekinSimplex, 2022-06-30:
Just a simple question.

Is there a way to include the output of a script in the index.mu which itself is not a script?

markqvist, 2022-06-30:
No, unfortunately there is no way to do this, an least not internally, using only nomadnet. if you need dynamic content in the index.mu, it must be a script.

But you could get some other job to dynamically generate the index.mu every time a change is needed, for example.