markqvist/NomadNet #8: [BUG] Terminal Colors Modified Upon Exit

tonydero, 2022-07-09:
I've never seen this before but here is what my terminal colors look like normally:
[image]
And this is what they look like after exiting nomadnet via Ctrl-q:
[image]
I know I can just exit and create a new terminal, but I assume that's not desired behavior, so I thought I'd report it.

markqvist, 2022-07-09:
Interesting, thanks for reporting this. Did you set nomadnet to 24-bit color mode in the config, or is this with the default 256-color mode? Does the colors return to normal if you run the `reset` command?

markqvist, 2022-07-09:
Also, would you mind sharing what terminal emulator and shell you are using?

tonydero, 2022-07-09:
Default config running in zsh 5.9 on wezterm 20210929-205847-2820b2c3. Command `reset` does work to reset the colors as well.

markqvist, 2022-07-09:
Thanks! I should bebable to figure out what is going on from that.

markqvist, 2022-09-13:
I am sorry, but I am completely unable to reproduce this using `WezTerm-20220905-102802`. I will close the issue now, but feel free to reopen if you are still having the issue or can provide any more clues on how to reproduce it.

markqvist, 2022-09-28:
After further investigation, I was able to reproduce it in other terminals.

closed 2022-09-28 by markqvist

markqvist, 2022-09-28:
This issue is now fixed. For future reference, Nomad Network handles terminal colors as follows:

- If nomadnet is configured for 16 or 88 colors, or the monochrome colo mode, it will set its own terminal color palette based on the colors defined in the user-configured theme. This is done to avoid situations where custom terminal colorschemes makes text unreadable. When exiting nomadnet, the program will restore the users normal color palette by writing an escape sequence that should be supported in most terminals.
- If nomadnet is configured for one of the high-color modes (256 colors or 24-bit color), it will not touch the terminal palette, but instead directly output color escape sequences for all characters, directly controlling the colors. As such no reset is needed upon exit.

commit 4da00e57 (2022-09-28): Better terminal palette handling. Fixes #8.