markqvist/NomadNet #10: Vim kyebinding

yagarea, 2022-07-10:
It is missed opportunity to not implement vim keybindings. The target group of this software will definitely find it useful.

markqvist, 2022-07-10:
Sounds like it could be useful. Can you elaborate on what keybindings you would like to see, and for what functions? Just the normal navigation bindings for moving around the program and text fields, or anything specific?

klundry, 2022-08-19:
> Sounds like it could be useful. Can you elaborate on what keybindings you would like to see, and for what functions? Just the normal navigation bindings for moving around the program and text fields, or anything specific?

That would be a good start. "h j k l" keys for left, down, up, right to navigate between windows and fields instead of having to move your hand over to the arrow keys.

markqvist, 2022-09-13:
Maybe I am being dense here, but when I was looking at implementing this, I can't figure out what the best way to handle input fields is when using "vim navigation mode". Every time you move over a text field, the `h j k l` keys would get interpreted as text input to those fields. The only way I can see to mitigate this is to add a global navigation/input mode to the program, as in vim itself. This would be overly confusing to new users, that are not familiar with how vim works. Unless there is some brilliant solution I have missed, I am hesitant to implement this.

klundry, 2022-09-14:
Hmm, yeah with text entry fields you would probably have to do something like elinks/w3m where you hit enter while having a field selected to activate entering text into that field. The only other way I could think of is using a modifier key to do the h j k l navigation. By holding ctrl or alt maybe.

markqvist, 2022-09-14:
The enter method is nice, but it is still going to require a global modeswitch of some sort to not confuse other users who are expecting it to work the other way around. I'll give the modifier key a thought, but I think it might be a bit clunky. All of this should maybe be added as a configuration option, that would have to be turned on by the user.

klundry, 2022-09-14:
I agree, having this as something you manually enable in the config would be fine.

markqvist, 2022-09-28:
Looking at this again, it seems the right way to go about it is to add the ability to map custom keys to nomadnet. Everyone is going to want something slightly different, so I think the only real solution is a general-purpose key mapping system through the config file.

Due to all the other work that is currently going on, this is not something I prioritise highly, and I will not work on it for the time being.

If anyone sees this, and wants to give it a shot, please go ahead, and feel free to ask any questions along the way :)

4c3e, 2023-02-15:
I ran into [ViEdit](https://github.com/laktak/urwid_viedit), which is a urwid text input field that supports vi-like controls.