markqvist/NomadNet #11: Openning editor does not work

yagarea, 2022-07-10:
I am using this software in alacritty and when I click `< Open editor >` in config tab, it blinks and nothing happens.

I have set neovim as my default editor.

markqvist, 2022-07-10:
Hmm, that is definitely not intended. By default, it will try to open what is defined in $EDITOR. I'll try running it with neovim set, and see what happens.

klundry, 2022-08-19:
It doesn't work for me either. When leaving the config as "editor = editor" it just blinks and doesn't open anything. If I change it to "editor = micro" It will open the config in micro but it's like it doesn't release control of the keyboard input and doesn't let me move the cursor or type and edit anything.

frankenstein91, 2022-08-26:
It does not work in the docker

markqvist, 2022-09-13:
> It will open the config in micro but it's like it doesn't release control of the keyboard input and doesn't let me move the cursor or type and edit anything.

Regarding this, it seems to be a bug in Urwid, but I am not quite sure yet. It is probably only the arrow keys that are not working, so try hitting backspace or a letter once, and the arrow keys should be usable too.

> It does not work in the docker

Since this is likely to be something completely different, this should probably be opened as a separate issue, along with log output and some steps to reproduce it.

> I am using this software in alacritty and when I click < Open editor > in config tab, it blinks and nothing happens.

I'm sorry, but I just cannot reproduce this. For me, using neovim works fine on a variety of different operating systems (also using Alacritty as terminal emulator). Could it be something on your end like a misconfigured alias or maybe just the editor directive being set to `neovim` instead of `nvim`, or similar?

Either way, I will close this for now as I am not able to make it happen here, and therefore unable to debug it.

closed 2022-09-13 by markqvist

srem1, 2023-02-01:
My too.
not happened whe I use editor

is possible add key bind QUIT?

markqvist, 2023-02-02:
> My too. not happened whe I use editor

I'm sorry, I don't understand. Can you be more specific?

> is possible add key bind QUIT?

Ctrl-Q is already mapped to quit.