markqvist/NomadNet #12: Use standard config location

yagarea, 2022-07-10:
Right now is config file in `~/.nomadnetwork/config`. It would be nice to add ability to  move it to `~/.config/nomadnetwork/config` or `/etc/nomadnetwork/config`.

In standard way software looks for config in this order:
1. `~/.config/nomadnetwork/config`
2. `~/.nomadnetwork/config`
3. `/etc/nomadnetwork/config`
4. loads defaults

It would be nice to implement this order so it will stop cluttering home dir.

markqvist, 2022-07-10:
This is a good suggestion, and I actually already thought about this a while ago, but I did not know what the standard way to go about it was, so I was hesitant to change it from ~/.nomadnetwork.

What do you think is the best default behaviour when no configdir exists? Create it under ~/.config if that folder exists, if not directly under ~/?

yagarea, 2022-07-10:
> Create it under ~/.config if that folder exists, if not directly under ~/?
That sounds reasonable. I think that is nice solution.

I am no expert but I think there is some convention using `XDG_CONFIG_HOME` env. variable. Maybe this would be cleaner solution.

markqvist, 2022-07-10:
> I think there is some convention using XDG_CONFIG_HOME env. variable. 

Thanks, I'll look more into this before implementing it. Thanks for your feedback and suggestions.

HarlekinSimplex, 2022-07-14:
Though it would require quite some changes to my bsbdock/nexus container configuration and usage approach, I would strongly support this change. It would clarify, simplify, cleanup and unclutter lots of things on my side too.
If no config is found and ~/.config does not exist yet, I would like to see RNS and Nomadnet creating ~/.config and beneath this folder their repective config homes ~/.config/reticulum and ~/.config/nomadnetwork.

commit 07f6bb83 (2022-09-14): Place configuration in .config dir by default. Closes #12.
closed 2022-09-14 by markqvist