markqvist/NomadNet #13: Nomadnet does not save 'saved' Servers regularily

HarlekinSimplex, 2022-07-14:
Don't know if this just happens on my side only.
But as soon as the nomadnet process is killed by server shutdown oder container shutdown, all 'saved' node definitions considered saved with the UI button '< Save >' are usually gone when the app comes up again.
Same 'usually' happens with the announcement stream.

'Usually' means that sometime those data are persisted as expected.
In the earlier versions of nn I was able to save those data by using 'Quit' on the UI and restarting nn again.
The actual 0.2.1 with rns 0.3.11 seem not to do so any longer every time I press 'Quit'.

I would love to see nomadnet writing the saved nodes list to disk everytime I press 'Save' after editing node details.
To persist the actual annoucement stream every time a new announcement arrives would be a nice thing too.
The latter is important to not miss any incoming announcement that may be saved as a saved node later.

Same with the demon mode of nn.
It looks like running nn as demon and then starting up another nn client with GUI does not deliver the announcement stream seen by the runing demon instance to the newly started nn client.
So having nn or rns persisting the announcement stream for any (nn) client to see would be a nice thing as well.
Maybe that was intent anyhow because sometimes it is showing up as expected, but most of the time it is not.

Both, the saved nodes list and the announcement stream are showing quite an unpredictable behaviour with regard to persistence on my side. As long as I have no clue how these lists should behave, I can't tell if this is a bug or not.

I run rns and nn in demon mode prestarted in the background.
Additionally I launch NN with the exact same config and storage folder the demon is using.

pls adv.

Cheers
Stephan

markqvist, 2022-07-14:
This may be a regression that occurred at some point. The intended behaviour is definitely for this data to persist as it is changed. I will look into it. Does it only happen for you when running nn in a container, or also when running directly in the host os?

HarlekinSimplex, 2022-07-14:
It was the same for host only usage as well since I started using nn 8 month ago.

The saved nodes were never saved in case the nn process was killed.

Guess is kind of a buffer flush or like that is missing to persist data that is  kept in memory intentionally still instead of saving it to disk immediately by the os.

HarlekinSimplex, 2022-07-15:
Think I found q workaround.

Click through the menus back and forth a couple of times and the pushing Quit seem to trigger the save for the node list eventually.

Just a hint that may help to narrow down the issue.

markqvist, 2022-09-13:
Fixed in ebfa6484a28626e0345705463a6a130d2d6f6a59 :)

closed 2022-09-13 by markqvist