markqvist/NomadNet #15: error: failed to prepare thin LTO module: Opaque pointers are only supported in -opaque-pointers mode (Producer: 'LLVM15.0.2' Reader: 'LLVM 15.0.2')

LaneaLucy, 2022-10-19:
Don't know what to do. Trying to install nomadnet on Android.
Getting
error: failed to prepare thin LTO module: Opaque pointers are only supported in -opaque-pointers mode (Producer: 'LLVM15.0.2' Reader: 'LLVM 15.0.2')

When doing pip3 install cryptography
Did everything from https://markqvist.github.io/Reticulum/manual/gettingstartedfast.html#reticulum-on-android

markqvist, 2022-10-20:
Hmm, that's a new one. What architecture is this Android device on?

LaneaLucy, 2022-10-20:
Output of uname -a
Linux localhost 5.4.147-qgki-24879461-abF926BXXS1CVHF #1 SMP PREEMPT Tue Aug 30 07:21:14 KST 2022 aarch64 Android

It's a Samsung z fold 3

LaneaLucy, 2022-10-20:
And it looks like it's the cryptography build, that throws this error

markqvist, 2022-10-20:
Thanks, great with the uname output.

Yeah, it's basically always `cryptography` that causes issues. Their build system has gotten a bit convoluted... But I wonder if it is anything specific to the Z Fold or just a generic thing that happens now because *something* was updated *somewhere*, which then broke build compatibility in Termux.

I'll look into it and see if I can reproduce (and fix) it somehow.

LaneaLucy, 2022-10-20:
I think I have solved the problem by installing python-cryptography

LaneaLucy, 2022-10-20:
So installing using pip failed because of the compiling error. But by installing cryptography using the python-cryptography with pkg worked, and now everything is working. Maybe you can add this to the nomadnet on Android manual section

markqvist, 2022-10-20:
Thanks! That is great to hear. I am going to update the intsall instructions with that. When I wrote the instructions, there was no precompiled package available, so this makes it much easier.

closed 2022-10-20 by LaneaLucy