markqvist/NomadNet #20: Bug: Nasty hang on macOS

acehoss, 2023-02-02:
I've been experiencing hangs on exiting nomadnet on macOS when the Auto Interface is enabled. It does eventually exit, and the duration of the hang seems to have a direct relationship with the duration of the nomadnet session.

Worst of all, during this hang, the macOS networking stack seems to freeze. Browsers, ssh sessions, everything stops communicating until the issue clears. Fortunately, disabling the Auto Interface cleared it right up. 

One other thing to note is I often have both Wi-Fi and ethernet connected to the same network. I've not had any problems previously, as the OS uses interface priority to pick which interface to use. But with broadcasting I can understand that might create unique challenges.

I'll try to collect more data.

markqvist, 2023-02-02:
Wow, that is definitely... Interesting. And not supposed to happen. How it's even possible to hang the entire IP stack from a userland process is quite intruiging. Thanks for investigating. If you can come up with some ideas on how to reproduce it reliably, I can probably also dig up a macOS machine to test it on.

acehoss, 2023-02-14:
Not surprisingly, this is not NomadNet-specific, but with the Reticulum stack in general. I've crashed my development computer a couple times leaving a test instance of the rnsh listener running for a few hours with the auto interface enabled. Everything's fine until I terminate the process, then the hard drive thrashes for a while and the computer freezes and reboots. Seems to depend on how long the process runs for.

I suspect this is related to software running on this particular computer. I run a NomadNet node on another machine with the same architecture and OS version and I've never experienced the problem on that machine. I'll keep gathering data.

markqvist, 2023-02-14:
Hmm, that is really interesting. I really wonder what is up with this. Thanks for investigating it so thouroughly.

markqvist, 2024-02-29:
Closing as fixed in subsequent release.

closed 2024-02-29 by markqvist