markqvist/NomadNet #22: ctrl-e closed the program, why?

srem1, 2023-02-08:
$ nomadnet --version
Nomad Network Client 0.3.2

fedora 37

markqvist, 2023-02-09:
Hmm, that is interesting. Just tried in 5 different terminal emulators. Cannot replicate here. What terminal emulator are you using?

Erethon, 2023-02-09:
I can confirm this is happening on `urxvt`, but only when there are no conversations/addresses to select from. The exit code for nomadnet then is `1`.

markqvist, 2023-02-09:
Hmm, interesting. Thanks for confirming in. It should be catchable then. Can't reproduce on `urxvt` here though, strange. I'll try to see if I can catch that keycode somewhere before it gets interpreted by the terminal itself.

markqvist, 2023-02-09:
Ok, it's in 2d6cd8e33fc9f25a997424cfdf1427bad4d577a4 if anyone wants to try it out :) Hopefully that works.

Erethon, 2023-02-09:
> Ok, it's in [2d6cd8e](https://github.com/markqvist/NomadNet/commit/2d6cd8e33fc9f25a997424cfdf1427bad4d577a4) if anyone wants to try it out :) Hopefully that works.

This didn't fix it for me :/

[This is my attempt at fixing it](https://github.com/Erethon/NomadNet/commit/4ec1a4979f1ffd95d436ee201b8ccf75d6fb854a), but I'm missing something obvious as it now crashes when it shouldn't (but it doesn't crash anymore when the list is empty).

Same problem happens when trying to delete a conversation and there are none.

Erethon, 2023-02-09:
> [This is my attempt at fixing it](https://github.com/Erethon/NomadNet/commit/4ec1a4979f1ffd95d436ee201b8ccf75d6fb854a), but I'm missing something obvious as it now crashes when it shouldn't (but it doesn't crash anymore when the list is empty).

This was me not applying the patch correctly /facepalm. If it looks good to you @markqvist I'm happy to open a PR for this.

markqvist, 2023-02-09:
Ah, I completely misdiagnosed it! It is indeed a bug that occurs *only* when the lists are empty ;) I somehow missed that completely.

The fix looks good. If you open the PR I'll get it merged!

closed 2023-02-09 by markqvist

Erethon, 2023-02-09:
@srem1 There hasn't been a release of NomadNet with the fix included yet. You can download this repo (`git clone` or download the zip file) and run `pip install .` to see if this fixes it for you.

srem1, 2023-02-09:
@Erethon That's me waiting for a fix ;)