markqvist/NomadNet #24: Bug: Trusted nodes reset every few power-offs

Neo-J-A, 2023-03-14:
Every few shutdowns my list of trusted nodes is reset.
This has happened to conversations as well once, but to trusted nodes twice.

The first time, I had added several propagation nodes. After powering off the computer and powering it back on, the nodes disappeared from my trusted list.

My OS is Debian 11 with Kicksecure.

markqvist, 2023-03-24:
Thanks for reporting this! When you say "shutdowns", do you mean shutting down the system, or just quitting the program?

I think there might be an issue where the program gets killed by the OS while it is in the middle of saving the data to disk, thus corrupting the files, and resetting them on the next program start.

But if this happens just with regular exits of the program, it is something else probably.

Neo-J-A, 2023-03-28:
Hmm, I forget if I made sure to close the program correctly first.

That sounds like a very logical explanation for it. I will test a few more times and see if I can reproduce it by killing the program.

markqvist, 2023-04-02:
Thanks! Let me know what you come up with! Either way, it may be a good idea to write to a temporary file on saving, and then quickly swap them afterwards, to avoid this issue.

markqvist, 2024-02-29:
Closing as fixed in subsequent release.

closed 2024-02-29 by markqvist