markqvist/NomadNet #25: NomadNet stops logging  after localnode disconnect/reconnect

blakangel, 2023-03-14:
Full Disclosure: I've just started experimenting w/ Reticulum, NomadNet and Rnodes in the last couple days and am still learning.

Using an Ubuntu machine with rnsd and nomadnet packages from yesterday.

I restarted my local rnsd this morning after switching Rnode hardware. Nomadnet client on same machinereported the LocalInterface socket was closed, then logged Reconnected after I restarted rnsd. No other log messages in the client nor in the file after that.

I'm also not seeing any new announcements in the Network tab but was able to announce my node and also create a connection with someone elses node.

blakangel, 2023-04-12:
Just a little update on nomadnet behavior I've observed when restarting rnsd:
The logging that stops is about propagation nodes:

> [2023-04-12 10:24:53] [Notice] Received active propagation node announce from <68fa4decab458c486086ae4db277015d>
> [2023-04-12 10:24:53] [Info] Selecting <cf9d43a6d6cfd63ff36a47558c24bd67> as default LXMF propagation node
> [2023-04-12 10:24:54] [Verbose] Link establishment timed out
> [2023-04-12 10:26:21] [Info] Selecting <cf9d43a6d6cfd63ff36a47558c24bd67> as default LXMF propagation node
> [2023-04-12 10:26:38] [Verbose] Link establishment timed out
> [2023-04-12 10:28:21] [Info] Selecting <cf9d43a6d6cfd63ff36a47558c24bd67> as default LXMF propagation node
> [2023-04-12 10:28:41] [Notice] Received active propagation node announce from <72d345df4d013383b4c48b6172826f2b>
> [2023-04-12 10:28:41] [Info] Selecting <cf9d43a6d6cfd63ff36a47558c24bd67> as default LXMF propagation node
> [2023-04-12 10:29:59] [Verbose] Initiating automatic LXMF sync
> [2023-04-12 10:31:40] [Verbose] Link establishment timed out

After it auto reconnects to local rnsd I do see logs of incoming links, LXMF syncs, etc. I can send direct messages and propagated ones (just setup an lxmd node for our group) but it looks like incoming announces are not processed and there is nothing new in the Announce Stream. I let it run for a few hours, re-announced my node/peer but no new announces coming in.

A restart of nomadnet gets the announces coming back in.

markqvist, 2024-01-03:
Fixed in RNS `0.6.9`, as far as I know.

closed 2024-01-03 by markqvist