markqvist/NomadNet #28: First Time Information and whole screen cannot be read - text is white-gray

qwertyuiopzxcvbnmlkjhgfdsa1, 2023-04-24:
Ubuntu 32.04, Gnome, run in docker as in instructions

I cannot read what is written there. Is the docker somehow missing capability of displaying color text? 

[image]

qwertyuiopzxcvbnmlkjhgfdsa1, 2023-04-28:
The wrong display colors were because of my system's setting. In Terminal's setting there is an option how to set it
[image]

closed 2023-04-28 by qwertyuiopzxcvbnmlkjhgfdsa1

markqvist, 2023-05-02:
For future reference, there is an option in the nomadnet config file an `~/.nomadnetwork/config` to set the color theme between light and dark. Most terminals use dark themes nowadays, so that is the default, but if you use a light theme, you can change the nomadnet theme to match.