markqvist/NomadNet #35: Unhandled exception trying to save a propagation node hash on announce stream page

blakangel, 2023-09-27:
In version Client 0.3.8 I tried pressing Ctrl-S on a propagation node and got the following:

> [2023-09-27 16:01:06] [Error] An unhandled exception occurred, the details of which will be dumped below
> [2023-09-27 16:01:06] [Error] Type  : <class 'TypeError'>
> [2023-09-27 16:01:06] [Error] Value : 'NoneType' object is not iterable
> [2023-09-27 16:01:06] [Error] Trace :
>   File "/home/bangel/.local/bin/nomadnet", line 8, in <module>
>     sys.exit(main())
>   File "/home/bangel/.local/lib/python3.10/site-packages/nomadnet/nomadnet.py", line 51, in main
>     program_setup(configarg, rnsconfigarg, daemon, console)
>   File "/home/bangel/.local/lib/python3.10/site-packages/nomadnet/nomadnet.py", line 11, in program_setup
>     app = nomadnet.NomadNetworkApp(
>   File "/home/bangel/.local/lib/python3.10/site-packages/nomadnet/NomadNetworkApp.py", line 363, in __init__
>     nomadnet.ui.spawn(self.uimode)
>   File "/home/bangel/.local/lib/python3.10/site-packages/nomadnet/ui/__init__.py", line 29, in spawn
>     return TextUI()
>   File "/home/bangel/.local/lib/python3.10/site-packages/nomadnet/ui/TextUI.py", line 211, in __init__
>     self.loop.run()
>   File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 287, in run
>     self._run()
>   File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 385, in _run
>     self.event_loop.run()
>   File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 790, in run
>     self._loop()
>   File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 827, in _loop
>     self._watch_files[fd]()
>   File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 416, in <lambda>
>     wrapper = lambda: self.parse_input(
>   File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 515, in parse_input
>     callback(processed, processed_codes)
>   File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 412, in _update
>     self.process_input(keys)
>   File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 513, in process_input
>     k = self._topmost_widget.keypress(self.screen_size, k)
>   File "/home/bangel/.local/lib/python3.10/site-packages/nomadnet/ui/textui/Main.py", line 83, in keypress
>     return super(MainFrame, self).keypress(size, key)
>   File "/usr/lib/python3/dist-packages/urwid/container.py", line 1135, in keypress
>     return self.body.keypress( (maxcol, remaining), key )
>   File "/usr/lib/python3/dist-packages/urwid/container.py", line 2316, in keypress
>     key = w.keypress((mc,) + size[1:], key)
>   File "/home/bangel/.local/lib/python3.10/site-packages/nomadnet/ui/textui/Network.py", line 1419, in keypress
>     self.parent.browser.save_node_dialog()
>   File "/home/bangel/.local/lib/python3.10/site-packages/nomadnet/ui/textui/Browser.py", line 665, in save_node_dialog
>     urwid.Text("Save connected node"+disp_str+" "+RNS.prettyhexrep(self.destination_hash)+" to Known Nodes?\n"),
>   File "/home/bangel/.local/lib/python3.10/site-packages/RNS/__init__.py", line 160, in prettyhexrep
>     hexrep = "<"+delimiter.join("{:02x}".format(c) for c in data)+">"
>

markqvist, 2023-09-28:
Thanks for reporting and including error logs! Do you happen to have the address of the node in question too? Might make it easier for me to replicate.

blakangel, 2023-09-28:
I tried to replicate it for awhile and can't cause the exception again.

zhabba, 2023-12-31:
Version: nomadnet==0.4.2

The same happens when trying to save **peer node** on _Announce stream_ using <C+s>.
Error log:
```
[2023-12-31 01:09:03] [Error] An unhandled exception occurred, the details of which will be dumped below
[2023-12-31 01:09:03] [Error] Type  : <class 'TypeError'>
[2023-12-31 01:09:03] [Error] Value : 'NoneType' object is not iterable
[2023-12-31 01:09:03] [Error] Trace : 
  File "/usr/bin/nomadnet", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.11/site-packages/nomadnet/nomadnet.py", line 51, in main
    program_setup(configarg, rnsconfigarg, daemon, console)
  File "/usr/lib/python3.11/site-packages/nomadnet/nomadnet.py", line 11, in program_setup
    app = nomadnet.NomadNetworkApp(
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/nomadnet/NomadNetworkApp.py", line 376, in __init__
    nomadnet.ui.spawn(self.uimode)
  File "/usr/lib/python3.11/site-packages/nomadnet/ui/__init__.py", line 29, in spawn
    return TextUI()
           ^^^^^^^^
  File "/usr/lib/python3.11/site-packages/nomadnet/ui/TextUI.py", line 211, in __init__
    self.loop.run()
  File "/usr/lib/python3.11/site-packages/urwid/main_loop.py", line 287, in run
    self._run()
  File "/usr/lib/python3.11/site-packages/urwid/main_loop.py", line 385, in _run
    self.event_loop.run()
  File "/usr/lib/python3.11/site-packages/urwid/main_loop.py", line 790, in run
    self._loop()
  File "/usr/lib/python3.11/site-packages/urwid/main_loop.py", line 827, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3.11/site-packages/urwid/raw_display.py", line 416, in <lambda>
    wrapper = lambda: self.parse_input(
                      ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/urwid/raw_display.py", line 515, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3.11/site-packages/urwid/main_loop.py", line 412, in _update
    self.process_input(keys)
  File "/usr/lib/python3.11/site-packages/urwid/main_loop.py", line 513, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/nomadnet/ui/textui/Main.py", line 83, in keypress
    return super(MainFrame, self).keypress(size, key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/urwid/container.py", line 1135, in keypress
    return self.body.keypress( (maxcol, remaining), key )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/urwid/container.py", line 2316, in keypress
    key = w.keypress((mc,) + size[1:], key)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/nomadnet/ui/textui/Network.py", line 1459, in keypress
    self.parent.browser.save_node_dialog()
  File "/usr/lib/python3.11/site-packages/nomadnet/ui/textui/Browser.py", line 665, in save_node_dialog
    urwid.Text("Save connected node"+disp_str+" "+RNS.prettyhexrep(self.destination_hash)+" to Known Nodes?\n"),
                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/RNS/__init__.py", line 161, in prettyhexrep
    hexrep = "<"+delimiter.join("{:02x}".format(c) for c in data)+">"
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

commit 6026f42f (2024-02-29): Check browser destination before displaying save dialog. Fixes #35.
closed 2024-02-29 by markqvist