markqvist/NomadNet #43: NomadNet GUI App Concept

RageGamerBoi, 2023-12-15:
I made a concept that's basically just a stripped down Firefox UI, but it seems to work surprisingly well.

[image]
[image]

The hamburger menu in the top right and the right click menu would be the replacement for the keyboard shortcut bar a the bottom of the screen (keyboard shortcuts and navigation could still work fine as well).

Hope this at least gives you some inspiration if you ever decide to make a NomadNet GUI.

RageGamerBoi, 2023-12-15:
Maybe a Thunderbird-esque design would work better :thinking:

markqvist, 2023-12-15:
Very nice, this is looking great!

At least for any foreseeable future, I will not be the one working on actually implementing it though, since I have to prioritize other parts, but I hope somebody else will take it up! Would be very cool to see something like this!