markqvist/NomadNet #54: Nomadnet shuts down after large message transmit/recieve, possible python issue?

skatesell, 2024-02-10:
Hi! I have two identical setups on Lenovo T430 laptops, with Nomadnet 0.4.6, Linux Mint 21.2 Cinnamon OS, with a LilyGO T3 v1.6.1 on 433MHz. The LoRa config is the default from the manual, with the frequency set to the 433 band.  

I have had no issues with short message conversations over LoRA. 

However I tried sending larger chunks of text, of larger than approximately 1000words, message is sent, but nomadnet shuts down. It restarts without issue, and displays the transmitted message. However it keeps intermittently shutting down when receiving messages, no matter how big payload there is on that message. This continues until the whole system is rebooted. This is the logfile output, it seems to involve python:  

`[2024-02-10 12:45:29] [Error] An unhandled exception occurred, the details of which will be dumped below
[2024-02-10 12:45:29] [Error] Type  : <class 'AttributeError'>
[2024-02-10 12:45:29] [Error] An error occurred in the external delivery callback for <LXMessage dc42dfffbc199edcd7bffe5262aeb1e705ba7babd0082fe466ee2bd87511a702>
[2024-02-10 12:45:29] [Error] Value : 'Text' object has no attribute '_cache_translation'
[2024-02-10 12:45:29] [Error] Trace : 
  File "/home/nomad/.local/bin/nomadnet", line 8, in <module>
    sys.exit(main())
  File "/home/nomad/.local/lib/python3.10/site-packages/nomadnet/nomadnet.py", line 51, in main
    program_setup(configarg, rnsconfigarg, daemon, console)
  File "/home/nomad/.local/lib/python3.10/site-packages/nomadnet/nomadnet.py", line 11, in program_setup
    app = nomadnet.NomadNetworkApp(
  File "/home/nomad/.local/lib/python3.10/site-packages/nomadnet/NomadNetworkApp.py", line 378, in __init__
    nomadnet.ui.spawn(self.uimode)
  File "/home/nomad/.local/lib/python3.10/site-packages/nomadnet/ui/__init__.py", line 29, in spawn
    return TextUI()
  File "/home/nomad/.local/lib/python3.10/site-packages/nomadnet/ui/TextUI.py", line 211, in __init__
    self.loop.run()
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/event_loop/main_loop.py", line 343, in run
    self._run()
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/event_loop/main_loop.py", line 445, in _run
    self.event_loop.run()
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/event_loop/select_loop.py", line 182, in run
    self._loop()
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/event_loop/select_loop.py", line 218, in _loop
    self._entering_idle()
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/event_loop/select_loop.py", line 171, in _entering_idle
    callback()
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/event_loop/main_loop.py", line 652, in entering_idle
    self.draw_screen()
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/event_loop/main_loop.py", line 669, in draw_screen
    canvas = self._topmost_widget.render(self.screen_size, focus=True)
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/widget.py", line 112, in cached_render
    canv = fn(self, size, focus=focus)
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/frame.py", line 408, in render
    body = self.body.render((maxcol, maxrow - ftrim - htrim), focus and self.focus_part == "body")
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/widget.py", line 112, in cached_render
    canv = fn(self, size, focus=focus)
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/columns.py", line 938, in render
    w.render(w_size, focus=focus and self.focus_position == i),
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/widget.py", line 112, in cached_render
    canv = fn(self, size, focus=focus)
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/widget.py", line 718, in render
    canv = get_delegate(self).render(size, focus=focus)
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/widget.py", line 112, in cached_render
    canv = fn(self, size, focus=focus)
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/widget.py", line 718, in render
    canv = get_delegate(self).render(size, focus=focus)
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/widget.py", line 112, in cached_render
    canv = fn(self, size, focus=focus)
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/pile.py", line 817, in render
    canv = w.render(w_size, focus=focus and item_focus)
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/widget.py", line 112, in cached_render
    canv = fn(self, size, focus=focus)
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/columns.py", line 938, in render
    w.render(w_size, focus=focus and self.focus_position == i),
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/widget.py", line 112, in cached_render
    canv = fn(self, size, focus=focus)
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/frame.py", line 408, in render
    body = self.body.render((maxcol, maxrow - ftrim - htrim), focus and self.focus_part == "body")
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/widget.py", line 112, in cached_render
    canv = fn(self, size, focus=focus)
  File "/home/nomad/.local/lib/python3.10/site-packages/nomadnet/vendor/additional_urwid_widgets/widgets/indicative_listbox.py", line 109, in render
    middle, top, bottom = self._listbox.calculate_visible(modified_size, focus=focus)
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/listbox.py", line 476, in calculate_visible
    p_rows = prev.rows((maxcol,))
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/widget.py", line 181, in cached_rows
    return fn(self, size, focus)
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/pile.py", line 852, in rows
    return sum(self.get_item_rows(size, focus))
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/pile.py", line 760, in get_item_rows
    rows_numbers.append(w.rows((maxcol,), focus=focused))
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/widget.py", line 181, in cached_rows
    return fn(self, size, focus)
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/text.py", line 281, in rows
    return len(self.get_line_translation(maxcol))
  File "/home/nomad/.local/lib/python3.10/site-packages/urwid/widget/text.py", line 300, in get_line_translation
    return self._cache_translation
`

markqvist, 2024-02-10:
Thanks for posting the errors. This is a bug in the UI rendering for large messages, apparently. Probably introduced after the recent updates to Urwid, since rendering very long messages used to work fine.

markqvist, 2024-03-01:
I cannot replicate this. Have tried with messages up to ~1,300 lines and 55 kilobytes of text, and everything works as intended. Closing for now.

closed 2024-03-01 by markqvist