markqvist/NomadNet #61: copy/paste linux terminals

ashworthpayne, 2024-10-18:
Running nomadnet on a Mac- awesome.

Running nomadnet on linux- very unawesome. I can't copy paste. tried a couple different terms in a couple different linux. Please let me know if I'm overlooking a local setting. Tried on rasp and new Debian.

SigmaEnterprise, 2024-10-18:
Ctrl + shift + c

Ctrl + shift + v

SigmaEnterprise, 2024-10-18:
I'm trying to connect to another node let me know if you want to connect?

ashworthpayne, 2024-10-18:
I'm sorry! I wasn't clear at all. I can't select anything using my mouse in linux. This behavior is observed on several different terminal emulators over two different Debian-based distros (new Debian 12 and raspbian). 
On my Mac, I can right-click (or two-fingers the trackpad) to get a menu... words are automatically selected and I can chose to copy the selected word. Very slick. So fancy. But on linuxen, I can't select, click, choose or anything. I can click on the command menus at the top, and node links work, but I simply can't select a node address or identity, making it impossibly to copy/paste. Sorry again for not being clear.

cascafico, 2024-11-15:
> Running nomadnet on a Mac- awesome.
> 
> Running nomadnet on linux- very unawesome. I can't copy paste. tried a couple different terms in a couple different linux. Please let me know if I'm overlooking a local setting. Tried on rasp and new Debian.

Just keep shift pressed when selecting text, then to paste keep shift pressed and left mouse click. Hope it helps.

markqvist, 2024-11-15:
The terminal is in "interactive mode" when running nomadnet. This captures mouse events, meaning you can't select text *directly*. You can completely disable this behaviour with the `mouse_enabled` option in the config file, **but** the best option is to simply use whatever mechanism your particular terminal has for getting into copy mode. All terminals have one, but the key/mouse combo for doing it varies, so look up the documentation for the particular one you are using, and all should be good!

Moving this to the forum.