markqvist/NomadNet #71: Custom module use and management per page?

kevinelliott, 2025-03-29:
**Describe the Bug**
Unable to `import yaml` because `pyyaml` module not installed in Docker image, however, is specific to only some pages. How might we be able to more easily manage this (such as with `uv` or `requirements.txt`)?

**To Reproduce**
`import yaml` or any other custom module in a page that does not exist inside of the Docker image.

**Expected Behavior**

**Logs & Screenshots**

**System Information**
- OS and version
- Python version
- Program version

**Additional context**
Add any other context about the problem here.

RFnexus, 2025-04-04:
@kevinelliott 

Not an expert at Docker by any means, but if you're using the Docker image from the [Dockerfile](https://github.com/markqvist/NomadNet/blob/master/Dockerfile) included in this repo, consider adding it system-wide on line 3 through `python-yaml` or through pip on line 8 through `PyYAML`. Same goes for any other packages you want to use for Micron files running with the Python interpeterer.

kevinelliott, 2025-04-04:
> [@kevinelliott](https://github.com/kevinelliott)
>
> Not an expert at Docker by any means, but if you're using the Docker image from the [...]

I did exactly this a few days ago, however, not every system (running python locally outside of docker) can modify the global packages (namely, macOS). This is fine of course, because you can modify the docker image to have the library you need globally and then just manage a local dev version of it using uv/pip. Just a little brittle!

It does make me want to consider **NOT** using the docker image for "production", but I do prefer it for simplicity. So I was hoping we could figure out a way to more dynamically allow modifications to the python modules for the docker image approach without requiring a rebuild via a custom Dockerfile.