markqvist/NomadNet #75: Since update nomadnet

cotteux, 2025-05-22:
Since i upgrade to nomadnet 0.7.0
When I try to connect to a node or send a lxmf message, try to load but nothing happen.
In the log , I have these 2 errors

2025-05-22 18:12:06] [Error]    An error ocurred while validating link request proof on <f831fc41533aa5c7133448d304291e08>.                                                                                     │
│[2025-05-22 18:12:06] [Error]    The contained exception was: Invalid link mode 0 in link request proof                

I try another software Retibbs and I was able to connect and use the BBS. I don't think is the rns the problem since I upgrade it at the same time than nomadnet

**System Information**
- Linux MintOS 22.10
- Python version 3.12.3
- nomadnet 7
- rns 0.9.6
- lxmf 0.7.1

markqvist, 2025-05-22:
This error occurs because the node you are connecting to is running an old version of RNS, that doesn't support the new AES-256 link and packet modes, which [is now used by default](https://github.com/markqvist/Reticulum/releases/tag/0.9.6).

Support for these modes were added in [version 0.9.5](https://github.com/markqvist/Reticulum/releases/tag/0.9.5), but apparently a lot of people haven't upgraded yet.

The fix here is to update nomadnet, or at least just RNS on nodes running old versions.

cotteux, 2025-05-22:
Thanks, I upgrade my node The_Gate and now I can access it. waiting for the rest of the people to upgrade